will127534/OneInchEye #2: DSO Astrophotography

bebon901, 2023-09-08:
Hi, 
I recently came across your sensor, and I've always been a bit of a tinkerer with Raspberry Pi's and astrophotography. I was wondering, how do you think this'd perform on a telescope? 
I know what I'm doing with a proper camera, and I used the HQ Camera before but gave up since it's not big enough. Is there good driver control for Analogue gain, shutter speed etc?
What's the low light performance like in comparison to the HQ Camera?
Cheers, 
Ben

bebon901, 2023-09-08:
Actually can see there's AG support!

will127534, 2023-09-11:
Yeap, there is auto gain, shutter speed and gain control.
Fun fact:
[image]

I'm also using it on my telescope too, though I've only tried it for solar observation. The other one will be allsky camera.
Haven't got the chance to do a proper low light performance comparison though.