will127534/OneInchEye #3: Sourcing the image sensor?

tlalexander, 2023-09-16:
Hello! I am not seeing any source online for the sensor. Maybe I am just missing it. Do you have any suggestions?

will127534, 2023-09-17:
I haven't tried to find it on Aliexpress, but I think you might find one or two reseller there.

matescb, 2023-10-17:
I found some on Alibaba but didn't try
https://www.alibaba.com/product-detail/-100-Brand-New-Original-IMX283EQJ_1600673386520.html?spm=a2700.galleryofferlist.normal_offer.d_title.6f9c6931VhjPxr

tlalexander, 2023-10-17:
Thank you! I sent an inquiry.

tlalexander, 2023-10-17:
Okay those are $65 each, with $30 DHL shipping. I will look more at getting these base boards made, and consider building some.

tlalexander, 2023-10-24:
Oh that’s not the real price. The listing I found also said $1 but when you contact the vendor they are $65.

amits26, 2023-12-26:
@will127534 how did you managed to solder Sony IMX283 on your PCB ? how difficult it was ?

bob454522, 2024-01-15:
Im also having issues find the sensor (ie sony IMX 585), without buying an entire camera that is.
where do you all buy it from? (im in usa)  thanks

hyperphonic0, 2024-01-18:
@bob454522:

Try here: https://www.aliexpress.com/item/1005006320816093.html

Riner06, 2024-10-06:
@will127534 @tlalexander @matescb @DomML @bob454522 Do anybody know if the imx283EQJ sensor work for the oneincheye pcb? And as I understand it doesn't shoot color, is that correct? Kind regards aspiring engineer(:

will127534, 2025-02-05:
@Riner06 It is still color sensor and yes it will work on this board.