will127534/OneInchEye #4: Pi 5 Drivers

SamM-Springbok, 2023-12-01:
I have one of the OneInchEyes from the initial batch, but have not had a potentially compatible device until the Pi 5.

For the StarlightEye it looks like a driver for the Pi 5 is in development.
https://github.com/will127534/StarlightEye/issues/2

Does a similar driver need to be made for the IMX294? Or should the current driver and instructions work fine?

will127534, 2023-12-05:
Yeap, I'm still working on it.....

I've tested OneInchEye, StarlightEye and IMX294 board all shows some issues on RPI5, I'm still figuring out what is happening.

will127534, 2024-01-21:
[imx283-v4l2-driver](https://github.com/will127534/imx283-v4l2-driver)

I've updated the IMX283 driver here.

Isonoglu, 2024-03-05:
I am experiencing a similar issue as SamM-Springbok described initially regarding compatibility with the Pi 5. Have you resolved the problems with the OneInchEye on the RPi 5, or are there still issues to be aware of? Additionally, is the updated IMX283 driver applicable to this situation?

After completing the steps in the readme from this link: https://github.com/will127534/imx283-v4l2-driver how can I verify that my camera system works on RPi 5?

I went through Quick-Start-Guide  https://github.com/will127534/OneInchEye/wiki/OneInchEye-Quick-Start-Guide but when tried to execute the last command: rpicam-still -r -o test.jpg -f -t 0 an ERROR pops up stating: "rpicam-aps currently only supports the Raspberry Pi platforms..."

Are there are any major differences between OneInchEye V1.4 and OneInchEye V1.3? I have both, but don't have a CM4 board to test them.

Have you any information about the power consumption of the OneInchEye camera? I'm also trying to connect the OneInchEye to the FPGA with a 4-lane connector, but have some problems. Could the power consumption of the camera be the case?

will127534, 2024-03-06:
Hi @Isonoglu 

Yes, I have resolve the issues, I've actually updated the drivers and guide last week, now it should work if you follow the setup guide (I've also verified it is working on my RPI5).

For the error, have you restart your RPI5? I also got that error if I did not reboot after recompiling rpicam-apps, but after a reboot the message will goes away.

There is no difference in terms of the Sensor between v1.3 and v1.4, so they should be the same.

For the power consumption, I don't have that handy but I will say it shouldn't be a huge issue (I've run it with a 50cm length FPC cable), if you have other board that is not RPI manufactured, can you check if the enable pin on the FPC has asserted?  There is a enable signal on the FPC that will power up/down the sensor. Enable pin is also one of the difference between V1.3 and V1.4, on V1.4 you can solder the enable pin jumper to ignore that enable pin or default to pulled-high.

will127534, 2024-03-09:
@Isonoglu Can you try re-install the v4l2 driver? I noticed that during the libcamera compiling, some package installed will trigger update-initramfs which then will caused dkms driver to disappear.

Isonoglu, 2024-03-10:
Hello, I just want to inform you that I went through the procedure in the Start Guide again and tried a different camera port, which resulted in success. You just need to change this line: 'sudo nano /boot/config.txt', because the config file isn't on that path. But everything else is straightforward."

will127534, 2024-03-11:
@Isonoglu Thanks for the confirmation!
Also note that you can add `,cam0 ` to the dtoverlay to move the port to Cam 0 (By default it is using Cam 1 because historical reasons: RPI default uses cam0 interface because that is the only one with 4-lane MIPI).

I'll update the guide as your suggestion and also some ports indicator!

Thanks a lot !

will127534, 2024-03-24:
Updated the quick start guide

closed 2024-03-24 by will127534