will127534/OneInchEye #6: Upstream the IMX283 and IMX585

phodina, 2024-01-05:
Hi @will127534 ,
thanks for cool project! I intend to use the sensor as well :-)

If you want I can upstream the sensors into the Kicad library so they could be used also by other projects.

I made some modifications to the schematics and footprint. More will be definitely needed for the symbol.

In the end I can open a pull request here with the modifications so that this project will use the symbol and footprint from the official Kicad libraries.

https://gitlab.com/phodina/kicad-footprints/-/commit/1206c688dd991517be1584b2a080f78f604ae323
https://gitlab.com/phodina/kicad-symbols/-/commit/510736715b263a41a45da1ea4ede2d00370c0934

Also could you share a link for the datasheet for the sensors? It's needed for the review process

amits26, 2024-01-05:
@phodina how you are planning to use sensor board with CM4 ? other than CM4 official IO board there is no other breakout board which exposes 4 lane MIPI CSI interface for CM4. Also drivers won't support RPI5

will127534, 2024-01-06:
Thank you! 
Feel free to do modifications and upsteam!

The only issue is that... Even though I did not sign the NDAs to get the datasheet (i.e I've gathered from the interenet), I'm a little bit hesitant to share publicly.

phodina, 2024-01-06:
> @phodina how you are planning to use sensor board with CM4 ? other than CM4 official IO board there is no other breakout board which exposes 4 lane MIPI CSI interface for CM4. Also drivers won't support RPI5

Well there are other boards that expose the 4 lane MIPI CSI interface e.g. the Quartz64 Model A

https://wiki.pine64.org/wiki/Quartz64 (under Expansion Ports)
> CSI - CMOS Camera Interface, 4 lanes MiPi up to 8 mega pixel on model A, 2 lanes MiPi up to 5 mega pixel on model B

I also work with the sensors in my day job and there are platforms such as HyIMAGE Pro V5U R2 IMAGE Toolkit that are for calibrating/testing of the image sensors.

https://cn.hyvision.co.kr/shop_view/?idx=12/#prod_detail_body

phodina, 2024-01-06:
> Thank you! Feel free to do modifications and upsteam!
> 
> The only issue is that... Even though I did not sign the NDAs to get the datasheet (i.e I've gathered from the interenet), I'm a little bit hesitant to share publicly.

Sure. As with any other image sensor it's always difficult to work with the manufacturers as they keep the lid on the specification and require to buy large volume just to give away some datasheets under NDA.

However, would it be possible to release at least the specification for the package (footprint) and the pinout?

Without that it's hard to check the symbol and footprint.

There are other products with this sensor so might be possible to ask them if they would share the datasheet:

https://www.arducam.com/product/arducam-20mp-imx283-camera-module-with-m12-mount-lens-and-adapter-board-for-depthai/

https://www.framos.com/en/products/fsm-imx585-sensor-module-26378

phodina, 2024-01-08:
FIY @will127534 

https://gitlab.com/kicad/libraries/kicad-footprints/-/merge_requests/3403
https://gitlab.com/kicad/libraries/kicad-symbols/-/merge_requests/4186

will127534, 2025-02-05:
Close for old issue.

closed 2025-02-05 by will127534