will127534/OneInchEye #7: v1.4, EQJ vs. CQJ, JLCPCB ordering

theplanekid, 2024-01-27:
I keep missing the Tindie restocks, so I’m going to bite the bullet and build some sensors myself.
@will127534 would you consider updating the files to v1.4 as you’re selling on Tindie? Completely understand if not.
What’s the correct sensor to use, IMX283EQJ or CQJ? What’s the difference? Can’t seem to find an answer myself; here and files say EQJ, Tindie listing says CQJ, so I’m a bit confused.
Also are there any considerations for ordering the PCB other than the filled vias? Anything to keep in mind when soldering the sensor itself?

Thanks!

will127534, 2024-01-27:
Yeap, building one yourself will be the fastest way to get one haha. Highly recommend if you have soldered/build SMD PCBs before. Also at this sensor size you would want to customized it in some degree too.

I'll upload the V1.4 later, yes - Actually I'll close this issue once I get it uploaded.
As for IMX283EQJ or CQJ... Honestly I don't know what is the difference either, but I use IMX283EQJ-C, I'll update the Tindle link later.

Nothing special about the PCBs other then filled vias, so I think that's all you need to take care of when ordering. For soldering the sensor, if you see condensing in the sensor when you cool down the board after soldering, don't worry it will go away.
Typically I use heatgun and heat it from the bottom of the board (Components side) or a IR pre-heater, but I have used heatgun from the top without problems (i.e cracking the glass) with low temperature soldering paste.

theplanekid, 2024-01-27:
I’ll start sourcing the sensors while I wait for v1.4, and thanks for the info!
Oh also, what size IR filters should I be getting?

will127534, 2024-01-27:
20mm x 16.8mm

theplanekid, 2024-01-27:
Thanks again!

will127534, 2024-02-17:
Updated PCB files to V1.4

closed 2024-02-17 by will127534