will127534/OneInchEye #13: Quick Start Guide suggestion

theplanekid, 2024-03-31:
Hello!
I’m rather inexperienced with RPi, and especially the 5, so I spent a decent part of today trying to install libcamera and the driver. I was only successful upon a fresh reflash, uninstalling rpicam-apps, and only then going through the install process.
Thus, a possible suggestion for the quick start guide; tell users to uninstall rpicam-apps before installing your forked version?
A friend who’s more experienced tells me that installing under an alias is also an option, so perhaps do that instead?
Also, some of the commands are the one-click copy thing and others aren’t, but that’s a very minor nitpick.
Once again, just suggestions.
Thanks for the cool hardware/software and the development work you’ve put in!

will127534, 2024-03-31:
I don't think you will need to uninstall rpicam-apps, because after you follow the libcamera install process, the next one is to install rpicam-apps, which should overwriting the existing one. 
I did go through the quick start guide a couple of times myself but I don't think I ever need to uninstall rpicam-apps, what was the issue when you follow the guide?

I'll update the guide to pack some cmds, thanks for the advice!

theplanekid, 2024-03-31:
Hmm odd, I assumed it would overwrite but I also ran through multiple attempts of the install process, reflashing each time, and got the rpicam-apps only supports the raspberry pi platform or something along those lines.

will127534, 2024-03-31:
If you got the "rpicam-apps only supports the raspberry pi platform" error, that can actually mean the driver is not loaded correctly, check if you see the imx283 in dmesg by `dmesg | grep imx283`, you should see some message like:

```
[    3.328314] rp1-cfe 1f00110000.csi: found subdevice /axi/pcie@120000/rp1/i2c@88000/imx283@1a
[    4.265573] imx283 6-001a: Device found
[    4.265620] imx283 6-001a: Pixel Rate : 451968432
[    4.265627] imx283 6-001a: Setting default HBLANK : 81, VBLANK : 336 with PixelRate: 451968432
[    4.265644] rp1-cfe 1f00110000.csi: Using sensor imx283 6-001a for capture
```

theplanekid, 2024-04-03:
I tried on a fresh SD card and everything went through without a hitch so I don't know what I was doing wrong the first time.
Such is life. Thanks for the help!

will127534, 2024-04-03:
I think the error "only supports the raspberry pi platform" is really misleading, I have scratch my head for a while debugging that also, I'll add a debug section to the quick start guide.

will127534, 2024-08-04:
Alright - Quick start guide updated

closed 2024-08-04 by will127534