will127534/OneInchEye #17: Rpi5 Failed to start with OneInchEye camera attached

watchever, 2024-07-02:
Hi Will,
I have built the OneInchEye camera through JLCPCB and received two samples (wow they are very nice and sensor is so big). All files I use (Gerber, BOM, Parts placements) are from this repository. I have ordered the IMX283 sensor from chinese supplier who sent it directly to JLCPCB. Results sound good, as you can see on attached pictures.
However when I attached camera to my Rpi5, well... (sadly), the Rpi5 led stays red and nothing happens.
I have tested several things : change the ribbon cable, attached to Cam1, Cam0, test my two samples of OneInchEye, verify components (as far as I can, compare to your OneInchEye pictures), ...

When I disconnect the camera, Rpi5 starts correctly... 
Unfortunately there are no informations (log file or anything else), because Rpi5 just doesn't start at all when camera is connected (well led just stay red).

Sorry to be unable to bring you more informations at this stage, but do you have any suggestions to try to test the camera differently or do something to understand what's wrong ?

Thanks a lot for your help

[image]
[image]

watchever, 2024-07-02:
I also add the JLCPCB gerber 2D viewer. Maybe a polarity issue ?
[image]

watchever, 2024-07-02:
Ok think I have found myself issue. The engraved on 22pins port are a mess.... Don't know why, like if Gerber files were corrupted.

will127534, 2024-07-02:
I download the gerber straight from github and upload to jlcpcb and the render is perfect:
[image]

I don't know why you think the gerber is corrupted, literally zip it, upload it. Also I want to stress that if you are not familiar with PCB and or electrical engineer, please look else where.

will127534, 2024-07-02:
There are multiple issues on your PCB board by the way

closed 2024-07-02 by will127534

watchever, 2024-07-02:
Hi Will, not saying that your PCB file is corrupted. It is by comparing my PCB file with yours that I found issues. So what I mean is my PCB files were modified in unexpected way by EasyEda. I have download Kicad files in EasyEda and generate gerber files with poor rendering. Really don't know why.

watchever, 2024-07-02:
So if you find others issues in my PCB screenshot other than the wiring of the 22pins, please let me know.

watchever, 2024-07-02:
Will, I know I am not a star with PCB. I do my best to learn, but if you can give me some advices to correct my mistakes on my previous PCB order, I will be really grateful. Maybe you will be happy to know that I plan to use the OneInchEye module for an astronomy project  which already use a IMX462 module but I want to upgrade to the IMX283. Thanks in advance for your time.

watchever, 2024-07-03:
I confirm that importing the Kicad project in EasyEda generate erroneous wiring...
I'd rather state it for other users, than just saying like @will127534 that I should look elsewhere (useless reply).

[image]