will127534/OneInchEye #22: Issue of striped lines appearing in photos.

kitcheng, 2024-09-04:
Thank you very much to the developers; this is an amazing project.

When I take photos at the highest resolution (5568x3664), there are striped lines at the edges of the photos. Is this a known issue that is considered normal, or is it abnormal and potentially related to factors such as software or cable connections?

[image]

will127534, 2024-09-04:
Yes it is normal, those are extra pixels/black pixels at the edge that can be cropped out.
Specifically, the issue is that for some reason I can not cropped out those at the driver level, if I set the resolution to the supposedly correct sizes on RPI5 it will causes stride unmatching.

kitcheng, 2024-09-05:
Alright, I understand. I'll crop it at the application layer. Thanks again.