will127534/OneInchEye #34: About the Tuning File

kitcheng, 2025-02-11:
Hello, first of all, thank you for your efforts—this is a great project.

I noticed that you have provided a tuning file, but I feel that the auto white balance tends to make the colors appear too red. I would like to know how this tuning file was measured and obtained. Do you have any suggestions or insights to share?

will127534, 2025-02-12:
Quick question just to clarify, do you have IR cut filter installed?

kitcheng, 2025-02-12:
> Quick question just to clarify, do you have IR cut filter installed?

Yes, I added an 850nm IR filter to the lens.

will127534, 2025-02-12:
notes that recalibration (i.e generate tuning files for your optical system) won't help the color I think because 650nm~850nm is in IR range that the sensor will pickup.

kitcheng, 2025-02-13:
> That might be the reason, I'm using 400nm to 650nm pass filter
> 
> [image]

Okay, thank you very much! I’ll try changing the filter.

closed 2025-02-13 by kitcheng