will127534/OneInchEye #35: Actual max frame rate for each readout Mode

ingenico-zanetti, 2025-02-26:
Hello,

from both https://www.sony-semicon.com/files/62/pdf/p-13_IMX283CQJ_Flyer.pdf and the comment in your code (imx283.c, in static const struct imx283_mode supported_modes_10bit[]), the maximum frame rate for MODE 1A should be just above 30 fps.

Yet, when using libcamera-vid, the available mode have a much lower max frame rate:

    SRGGB10_CSI2P,5568x3094/21.3074 - Score: 0
    SRGGB10_CSI2P,5568x3664/17.9931 - Score: 4996.28
    SRGGB12_CSI2P,2784x1828/35.5454 - Score: 9929.91
    SRGGB12_CSI2P,5568x3664/17.9931 - Score: 5996.28

This does work up to 20 fps (excerpt from syslog):

[ 6434.987915] rp1-cfe 1f00128000.csi: Using a link rate of 1440 Mbps
[ 6435.422649] imx283 11-001a: Using clk freq 24 MHz
[ 6435.449600] imx283 11-001a: Mode: Size 5568 x 3094
[ 6435.449602] imx283 11-001a: Analogue Crop (in the mode) 40,393 5472x3078

Above this frame rate (30 fps), the 2784x1828 12bpp mode is selected (by application ? libcamera ? driver ?)

[ 7547.113476] rp1-cfe 1f00128000.csi: Using a link rate of 1440 Mbps
[ 7547.523729] imx283 11-001a: Using clk freq 24 MHz
[ 7547.550688] imx283 11-001a: Mode: Size 2784 x 1828
[ 7547.550691] imx283 11-001a: Analogue Crop (in the mode) 40,108 5472x3648

Has it something to do with the clock being only 24MHz and not the maximum 27MHz stated in the datasheet ?
Or a limitation on maximum pixel throughput on Pi 5 side ?
Is it be possible to have 4K at 30 fps with the IMX283 on the Raspberry Pi 5 ?

will127534, 2025-02-26:
https://www.willwhang.dev/Solar-eclipse-2024/#overclocking-the-rp1-for-enhanced-pisp-performance

closed 2025-02-26 by will127534

ingenico-zanetti, 2025-02-27:
Wonderful: it is now running at 30 fps in 5568 x 3094 10 bpp !
Thanks a lot.