will127534/OneInchEye #36: Driver installation issues

eva-00100, 2025-05-07:
I made the board according to the requirements, and now there is such an error after installing the driver I'm not sure if it's a problem with the board or a problem with the driver installation Please help me This problem has been bothering me for many days.

[image]

eva-00100, 2025-05-07:
This is a screenshot of the driver when it was installed.
[image]

will127534, 2025-05-07:
driver install looks good, so it is the problem of your board that the driver can not read the sensor

closed 2025-05-07 by will127534

eva-00100, 2025-05-07:
Thank you, I will try to solve this problem

eva-00100, 2025-05-08:
I am not very familiar with hardware, and the board is outsourced to a factory, so I would like to ask if there is a high probability of sensor damage😂😂

will127534, 2025-05-08:
I can't magically diagnose your hardware virtually, so I don't know.
What I noticed however, is your current driver is loaded for CAM0, make sure you have the camera module connects to the right connector.

eva-00100, 2025-05-08:
Thank you very much. I will continue to try to solve the problem❤️😃

eva-00100, 2025-05-08:
Maybe the problem was caused by me using the opposite FPC cable

eva-00100, 2025-05-08:
I reversed the FPC cables and it became like this.
[image]

eva-00100, 2025-05-08:
The 3.3v on the board is there, but the 1.8v and 2.9v my ammeter shows none.

will127534, 2025-05-08:
Can you pose a few (good) images of your board along with a few more showing how are you connecting the camera to RPI

eva-00100, 2025-05-08:
Of course, I'm sorry for replying late
[image]

[image]

eva-00100, 2025-05-08:
I understand what you mean, I downloaded the package on github, unzipped it, opened it with kicad, and then packaged it into a zip pcb made with JLCEDA

will127534, 2025-05-08:
I'm noticing a lot of difference between the board and what it should looks like, you will need to double check with JLCEDA on the correctness after converting that. In general, converting the PCB file format is like converting Python code to C code, while Gerber is the compiled output.

i.e if you compare the PCB image side by side with KiCad and converted result on JLCEDA, there should be quite a lot of differences and I think that is why.

eva-00100, 2025-05-08:
I understand, thank you very much for your help. I will continue to study and research this knowledge. Thank you for your help❤️❤️😃😃

eva-00100, 2025-05-20:
This time I made the circuit board according to the requirements and successfully recognized it on the Raspberry Pi, but I encountered the problem shown in the picture. I replaced the ribbon cable several times, but the error still occurred.

[image]
[image]

will127534, 2025-05-20:
No frame received = connection issue on the CSI-2 lanes. I2C device found = Power and I2C communication are good.

Check the CSI-2 Filter origination and the connection on the FPC connector.

eva-00100, 2025-05-20:
Thank you very much for your help, I found that the welding of FL5 when I welded the virtual soldering problem solved perfectly Thank you again for your help❤️😀😀