will127534/OneInchEye #37: two camera problem

almowery, 2025-06-06:
Will I have had great success with your sensor imx283. I wonder if you know of this problem with still pictures with raspberry pi.

starting with rpi 5 10 sec exposures I found on the web this script would work the best removing the one frame drop issue.  first exposure time 20sec ,second exposure 10 seconds.  The first exposure takes 2x the --shutter time.

rpicam-still --output test.jpg --timeout 1 --analoggain 1 --awbgains 2.51,2.1 --saturation 1 --contrast 1 --sharpness 1
 --brightness 0 --quality 100 --nopreview --width 4056 --height 3040 --shutter 10000000 --autofocus-mode manual --immediate
--camera 0/1
 
 The one frame drop would cause the time to get a 10sec exposure to increase to 20sec.  The typical camera using one frame 
 to adjust then one frame to capture.  with the oneincheye the first exposure takes 20sec then the second back to 10sec.
 
 I have been trying to get two cameras to work together and I always had trouble getting a second camera through Tindie, so I
 bought two from arducam.  Their connector is reversed and they seemed to work fine except I could never get them to drop
 exposure time down to 10sec.  even starts first frame at 40sec.  then next frame is 20sec.
 
 I dont see why the two boards could behave so different.  Here is a table of operation using arducam(ard) and William(will)
 (port, port,cam0 exposure first seconds, cam0 exposure second seconds, cam1 exposure first seconds, cam1 exposure second seconds)
 
 Cam0 	Cam1	cam0 first	cam0 second	cam1 first	cam1 second
 ard			        40			20
		ard								40			20		
		will		        					60			60
will			        20    		10
will	ard		        40     		20			20			10

It's weird by itself will's cam does good 20/10.  with two cameras will's cam does bad 40/20 and ard camera does good 20/10.
I have considered making a helperfile(compile in libcamera)  for imx283.  I read that through this one can stop the one frame drop, first change in exposure.
Do you have any suggestions for the difference between arducam and Will's camera and why the inconsistency with which camera has 20/10 first exp/second exp. 
I thought it was the csi-4 lane bus issue, but it has to be something to do with enumeration of each camera and what "helper" or "default" file is selected.  
40sec would indicate 3 frames dropped and one taken.  I don't even know of a camera that drops three frames.  Any ideas.  Have you seen this?

almowery, 2025-06-06:
first --camera 0 is Will camera
second --camera 1 is Arducam

notice when the camera starts line 8-21 both cameras are activated i2c@80000 and i2c@88000

jackblack@raspberrypi:~/star_recognizer3 $ rpicam-still -o test_data/testR.raw --timeout 1 --tuning-file imx283.json --analoggain 1 --awbgains 1.0,3.5 --saturation 1 --contrast 1 --sharpness 1 --brightness 0  --nopreview --autofocus-mode manual --immediate --gain 1 --shutter 10000000  --camera 0 -v
[1:17:47.363857394] [2523]  INFO Camera camera_manager.cpp:284 libcamera v0.0.0+4579-80845ad5
[1:17:47.365970068] [2524]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 07-03-2024 (09:04:18)
[1:17:47.367589833] [2524]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'imx283'
[1:17:47.367611685] [2524]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[1:17:47.374573967] [2524]  WARN RPiAlsc alsc.cpp:170 no luminance table - assume unity everywhere
[1:17:47.376080491] [2524]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx283@1a to CFE device /dev/media2 and ISP device /dev/media4 using PiSP variant BCM2712_C0
[1:17:47.376136547] [2524]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/pisp/rpi_apps.yaml'
[1:17:47.376292547] [2524]  INFO RPI pisp.cpp:662 libpisp version v1.0.4 6e3a53d137f4 07-03-2024 (09:04:18)
[1:17:47.377391403] [2524]  WARN CameraSensorProperties camera_sensor_properties.cpp:274 No static properties available for 'imx283'
[1:17:47.377424199] [2524]  WARN CameraSensorProperties camera_sensor_properties.cpp:276 Please consider updating the camera sensor properties database
[1:17:47.383102996] [2524]  WARN RPiAlsc alsc.cpp:170 no luminance table - assume unity everywhere
[1:17:47.384442408] [2524]  INFO RPI pisp.cpp:1121 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx283@1a to CFE device /dev/media1 and ISP device /dev/media3 using PiSP variant BCM2712_C0
[1:17:47.384487445] [2524]  INFO RPI pipeline_base.cpp:1144 Using configuration file '/usr/share/libcamera/pipeline/rpi/pisp/rpi_apps.yaml'
Options:
    verbose: 2
    info_text:
    timeout: 1ms
    width: 0
    height: 0
    output: test_data/testR.raw
    post_process_file: 
    preview: none
    qt-preview: 0
    transform: identity
    roi: all
    shutter: 10000000us
    gain: 1
    metering: centre
    exposure: normal
    ev: 0
    awb: auto
    awb gains: red 1 blue 3.5
    flush: false
    wrap: 0
    brightness: 0
    contrast: 1
    saturation: 1
    sharpness: 1
    framerate: 30
    denoise: auto
    viewfinder-width: 0
    viewfinder-height: 0
    tuning-file: imx283.json
    lores-width: 0
    lores-height: 0
    autofocus-mode: manual
    autofocus-range: normal
    autofocus-speed: normal
    autofocus-window: all
    hdr: off
    mode: unspecified
    viewfinder-mode: unspecified
    metadata: 
    metadata-format: json
    encoding: jpg
    quality: 93
    raw: 0
    restart: 0
    timelapse: 0ms
    framestart: 0
    datetime: 0
    timestamp: 0
    keypress: 0
    signal: 0
[... 160 lines cut ...]
    NoiseReductionMode : [0..4]
    AeConstraintMode : [0..3]
    FrameDurationLimits : [55577..164960255]
    AnalogueGain : [1.000000..22.505495]
    AeFlickerMode : [0..1]
    AwbMode : [0..7]
    AeMeteringMode : [0..3]
    AeExposureMode : [0..3]
Buffers allocated and mapped
Still capture setup complete
Requests created
Camera started!
Camera stopped!
Still capture image received
Exposure time: 9999083
Ag 1 Dg 1.00009 Total 1.00009
Thumbnail dimensions are 320 x 240
Thumbnail size 15175
JPEG size is 8845179
EXIF data len 298
Saved image 5472 x 3648 to file test_data/testR.raw
Closing RPiCam application(frames displayed 0, dropped 0)
Camera stopped!
Tearing down requests, buffers and configuration
Camera closed

will127534, 2025-06-06:
Which driver are you using for Arducam module?

almowery, 2025-06-06:
The only driver I have is the driver u created.  Arducam never furnished a driver for raspberry pi, they sold it for the Jetson.  They are not selling the module separate anymore they are selling a Pivistation 5 which has a preinstalled OS and I assume drivers but is sold as a video camera.

almowery, 2025-06-06:
This is the rpicam-apps thread about the helper files and the extra frame.

https://github.com/raspberrypi/rpicam-apps/issues/524

will127534, 2025-06-06:
Ok so, to summarize your question:
> they seemed to work fine except I could never get them to drop
> exposure time down to 10sec. even starts first frame at 40sec. then next frame is 20sec.

You want to capture one frame immediately (10s exposure -> 10s of total time spend on capturing), but you can't do that with 2 camera module attach?

almowery, 2025-06-06:
uc-769 discontinued, interesting that it works at all.

I found this page
https://www.arducam.com/arducam-20mp-imx283-camera-module-with-m12-mount-lens-and-adapter-board-for-depthai.html

https://github.com/luxonis/depthai-python/issues/725

I originally purchased through Uctronics
Arducam 20MP IMX283 Camera module with C mount lens for DepthAI OAK
SKU: B0301

Mine has a ir cut filter gizmo to move filter in/out.  You don't even see those on their advertisements.  

I know your driver works on them, but obviously not perfectly.  i just had such a trouble getting two from tindie.  I own one of yours from early on you development.  I saw someone was able to manufacture some using your files.  Is that really possible?  Possible to get the imx283? Thanks for fast responses.  AL

almowery, 2025-06-06:
https://www.fabtolab.com/arducam-b0301-20mp-imx283-camera-module-c-mount-lens-adapter-board-depthai-oak

almowery, 2025-06-06:
Your hint to check to see if the OSC is present worked out well. Thank You.  The OSC is 12.000Mhz

Ok so I found the method to stop the extra frame using the line in rpi-apps.yaml.  I determined the arducam does have the osc hooked up and the osc is 12.000Mhz and not the 24.00Mhz on your board.  That would explain the time that it takes for an exposure is 2x.  I can change the osc on the arducam boards but I thought I could test it with a software recompile, first.  I see in your imx283 driver that each of the typical clock speeds are present in an array.  I can't figure and the clk_get_rate() function get's the speed set by the system.  I just don't known where this comes from and how to change the setting so your driver loads the 12mhz values into the imx283 registers instead of the 24Mhz values.

Very close.  Thank you.  AL

will127534, 2025-06-07:
So I guess that is why, the PLL is setup as if it is a 24Mhz clock, I'll recommend you swap it to 24Mhz to align both of your camera. Or.... you just /2 on OAK camera module and have the same result I guess. (Not sure if that will impact other stuff but given that we are running IMX283 half the speed it should be fine)

closed 2026-02-17 by will127534