will127534/OneInchEye #38: Not booting with Pi 5

malcolmjay, 2025-06-08:
Hi Will,

I have the camera board connected into the cam 1 slot on the pi 5 and followed your fpc instructions. The pi doesn’t boot whenever I have the camera connected.  I’ve installed all of the drivers - any ideas on how to troubleshoot this?

Thank you,

Malcolm

malcolmjay, 2025-06-08:
Yes I’m using a single sided FPC - attached are photos.

[image]
[image]

will127534, 2025-06-08:
So.. no, this is double side FPC where you have contact on the different side.

This is the single side FPC:
[image]

malcolmjay, 2025-06-08:
Thank you! I will buy the correct cable.

closed 2026-02-17 by will127534