will127534/OneInchEye #42: More stock to buy/borrow?

Babyyoda777, 2025-08-26:
Hi, I’m working on a custom CM5 carrier PCB here at https://github.com/azlan-works/Argo and was looking to use the OneInchEye but unfortunately it’s completely out of stock. Is there a timeline for when there will be any available to get @will127534 or would it be possible to borrow one for testing in the meantime (would really appreciate any help!).

My carrier pcb is tiny and includes built in battery management and I’m trying to optimise for portable photography. This is the best all round native CSI board I’ve seen and can’t really source the sensor myself either. Would love it if we could work on this!

will127534, 2025-08-27:
Neet, I do have a few for my own projects so can definitely send one to you, send me a message to will [at] willwhang.com

Babyyoda777, 2025-08-29:
Hey @will127534 just had a side thing to add, came across this promising looking successor to the IMX283:

https://www.ovt.com/wp-content/uploads/2025/04/OV50X50-PB-v1.0-WEB.pdf

It says it's currently in sampling and then production in Q3 2025 but at 50MP and 1" it also supports pixel binning to 12.5MP which seems cool.